jgyates/genmon #1219: Add charger meter to show mv charger is pulling

grsthegreat, 2024-12-18:
I just noticed that Generac MobilLink wifi monitoring app now shows charger mA draw. This is helpful to see how onboard controller charger is working. Is there any way for genmon to see this value in the controller?

[image]

star-glider, 2024-12-19:
There's not a gauge for it in the dashboard, but it is presented in the text in the detailed "status" section of the main page:

[image]

jgyates, 2024-12-19:
The missing piece here is: it depends on what controller you have.

@star-glider, looks like you have an H-Panel or G-Panel

@grsthegreat you don't mentioned what controller you have. That info would be needed, along with your registers sumbitted on the about page in order to add this for Evolution controllers. 

At the moment the only data for carger current is displayed if you enabled "Display Unknown Sensors" on the settings page. The reason I put this on the under unknown sensors is that I did not have a way to validate this info and I was looking for some feedback from folks with generators with controllers other than the one that I have.

I am guessing this is only for Evolution 2.0 and Evo Liquid cooled.

jgyates, 2024-12-19:
@grsthegreat 

Specifically I am interested in register 05ee when it is charging and when it is not charging. On my controller this value never goes to zero, it goes 8 mA when not charging.

skipfire, 2024-12-19:
If it helps, I have an Evo 2 AC and it shows 60 with a range of 0 to 60.
[image]

grsthegreat, 2024-12-19:
I have evolution 1 controller. Status page only shows battery voltage on mine.

[image]

jgyates, 2024-12-19:
@grsthegreat, Is your generator hooked up to both genmon and mobiel link?

Look on the live register view page (double click the gear icon in the top right of the web interface once). Then search on 05ee. This will show the contents of modbus register. What is the contest of registr 05ee (like the screen cap that @skipfire sent).

grsthegreat, 2024-12-19:
[image]

jgyates, 2024-12-19:
What is the controller for the cell tower?

grsthegreat, 2024-12-19:
Evolution 1

jgyates, 2024-12-19:
I will update this thread when I have something to test. I am doing some testing now, just waiting for my battery charger to kick in.

jgyates, 2024-12-24:
I have added this as text on the status page. I am still waiting on a charging cycle so there may still be changes, but it is in the repository if someone wants to update their software and give any feedback. This is enabled for all Evolution models, but I am not 100 percent sure it works on liquid cooled units yet as my charging cycle has not kicked in yet.

Adding this as a gauge might come later. At the moment I don't think a gauge is going to add anything as this value will likely not change a lot. The controller will pick a charging voltage and go with that. This comment in the code has some more details on the battery charging:

https://github.com/jgyates/genmon/blob/f12af352403bc3537577325dccce32eca0b62152/genmonlib/generac_evolution.py#L4471

grsthegreat, 2024-12-24:
Will this require standard  update or is a copy and paste type of thing?

jgyates, 2024-12-24:
Standard update

grsthegreat, 2024-12-24:
Ah….i just updated, and now i see battery charger

grsthegreat, 2024-12-24:
But its jumping around alot, one second its 37mA, than its 390 mA, than 89 mA in a few seconds

jgyates, 2024-12-24:
@grsthegreat , what is your controller type?

grsthegreat, 2024-12-24:
Evolution 1 on a 6 year old 22kw gen

jgyates, 2024-12-24:
It is possible that Evo 1 does not update this value with any meaningful results. Currently genmon is just reading the modbus register and displaying the value.

jgyates, 2024-12-26:
My charging cycle finally kicked in and it appears to work as expected on a Evo liquid cooled variant.

Any other input on Evo 1? I wonder if it could firmware version dependant. @grsthegreat what is your firmware version?

grsthegreat, 2024-12-26:
V1.20

closed 2024-12-27 by jgyates

dennis48755, 2024-12-27:
Hello, sharing what I see as another Evolution 1.0 (firmware v1.22) owner.

Yesterday my Battery Charger Current was steady at 0 mA whether voltage was 12.4 or 13.1. 

Today Battery Voltage is 14.30 and the Battery Charger Current is jumping around; 0 mA, 9 mA, 25 mA, 203 mA, 231 mA, etc.

Dennis

grsthegreat, 2024-12-27:
Thats the same as mine. Jumps all over. Weird

jgyates, 2024-12-27:
I am assuming that the Mobile Link status that @grsthegreat is showing is updated at a lower frequency than genmon, which is likely the reason you don't see a the ML battery charger current jumping around.

I can take this out or leave it in, any thoughts from Evo1 users?

grsthegreat, 2024-12-27:
> I'm ok either way.

Leave it

jgyates, 2024-12-27:
ok. I will leave it. It probably will not, but if it causes confusion we can take it out later.