jgyates/genmon #1227: Power Output not displaying correctly during a Power Outage on Genmon

bbenton1341, 2024-12-31:
### Steps to Reproduce

During a utility power outage situation today, approximately 1 hour in duration, Genmon Power output shows approximately .6 KW, regardless of any large household load(s).  

### Expected Result

Relatively accurate Generator Power Output.

### Actual Result

.6KW consistently, +- .1KW
With generator running during a test with no transfer switch operation switch to generator power, the Power Output shows the same power output as when generator is under load for whole house.
With Generator off, Power Output does show 0KV.

### Version Info

Version V1.19.05
Generator: Generac 32KV, Evolution Controller.

### Logs

Logs and Registers submitted via the About page approximately today 1400 PST, during the power outage.

bbenton1341, 2024-12-31:
Additional note: The Status Page Output Voltage and Current text values are consistent with the gauge display of .6KW.

jgyates, 2024-12-31:
Hi @bbenton1341 

I received your logs. The liquid cooled Evolution models export the reading of the internal Hall Effect Sensor. This value is used in calculating the current for liquid cooled models. The short answer is that you may need to calibrate the controller readings to get an accurate result.

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues#5-current-and-power-output-values

You have a few options:

1) set a value of current offset and / or current divisor on the Advanced Settings page to make the value read from your controller match the current consumed by your house. The above link has some suggestions for those values.
2) disable the display of power on the Advanced Settings page by changing the "Disable Power / Current Display" value
3) You can import current values that genmon will use using the MQTT input add on, but this requires additional software and a meter to read the data

Let me know if you have any questions.

bbenton1341, 2024-12-31:
Thanks for your quick response.  I’ve read the link you provided below.

Question:

Could the default current offset / current divisor currently set explain why the values I’m seeing both with no load or large load are the same?

 

Bob

 

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues#5-current-and-power-output-values

jgyates, 2024-12-31:
possibly, it is hard to say without knowing the values of registers 0058 over the range of the load. You can view the value of register 0058 while your generator is powering a load from the live register view page (click the icon in the top right of the web interface once for the live register view, click it twice for advanced settings). This registers is divided by 1000 to get the value of the current. The current and voltage are used to calculate power. What some people have done is use a clamp on current meter and measure the current on each leg, and add the readings together. Then the note the value of registers 0058 and use a value of the current divisor and current offset settings to make displayed current reading match the clamp on meter reading. Unfortunately this is about the best you can get with the Liquid Cooled Evolution units. The air cooled untis (mostly) have built in current transformers. 

Here are some threads that discuss this issue: https://github.com/jgyates/genmon/issues?q=is%3Aissue+label%3APowerDisplay+Hall

bbenton1341, 2025-01-01:
Thanks.  You may close this case.  I’ll try the meter approach, and adjust the registers as needed and see where that gets me.  

If I get it resolved I’ll update the case, if allowed.

 

Bob

jgyates, 2025-01-01:
ok. Feel free to post any follow up questions or status to this thead.

closed 2025-01-01 by jgyates