jgyates/genmon #1240: i cannot get my genmon device to log into my network

grsthegreat, 2025-01-17:
### Steps to Reproduce

i rebooted the genmon device and i see genpi under available wifi.

i lig into this with password and it connects. i then enter website wifisetup.pi:1880/GenPi

i set the network name and password and hit select.
system logs out and genpi disappears from available wifi

but nothing appears in fing for the generator. in about 20 minutes i can see genpi again in available hotspots.

SO, i went and rebooted genmon again, logged onto it on desktop computer and was able to log onto it thru Putty

while in putty i typed
sudo raspi-config and enter

under second option i hit enter but it wont let me enter a network name and password, instead i get error as follows "rasperi pi could not communicate with wpa supplicant"

### Expected Result

not sure, i cant get into my network

### Actual Result

cant log into network, tried 10 times

### Version Info

pi3 model B Rev 1.3

### Logs

cant access

grsthegreat, 2025-01-17:
i also tried
cd genmon
sudo ./genmonmaint.sh -u
and entered Y
./startgenmon.sh restart

didnt help at all, still cant get onto network

jgyates, 2025-01-17:
@grsthegreat 

I would need more info before I would have any suggestions. You listed a lot of the things you tried, but you don't really give much info about what is an what is not working.

you did not show the output of the above commands.
you do not mention if you have email setup.
you did not mention if this was a new install or has anything changed recently
Was the install script run on this system?
Has it ever worked before on this system?
It is unclear what your root issue is: unable to connect to wifi or unable to load the genmon web interface or unable to load genmon.

grsthegreat, 2025-01-17:
sorry,
system has been operational for several years. i just noticed it wasnt alerting me when it exercised. when i tried to log into it, nothing happened. i have not changed anything. same wifi setup. same email setup.

when i ran the above script, it ran just fine. it got back to where it rebooted as it usually does.

when i reboot the genmon unit, i can see it as a hotspot. i can log onto the hotsopt and bring up website to set account and password

but after pressing setup, nothing happens.

thats why i went thru putty to try and establish link to my router and password, but putty just gives me this error

grsthegreat, 2025-01-17:
also, i updated firmware maybe 3-4 weeks ago and its been working great up until last wednesdays exercise

skipfire, 2025-01-17:
@grsthegreat how are you connecting with Putty?  A serial connection or by name/IP?

grsthegreat, 2025-01-17:
name/ip

skipfire, 2025-01-17:
Were you able to connect with putty before activating the hotspot?

grsthegreat, 2025-01-17:
its probably time to upgrade the pi device to a newer version, but i have no idea how to program the thing if i did that

grsthegreat, 2025-01-17:
i had to connect my computer to the genpi hotspot before i could communicate with the genmon unit. 

the unit itself ip address does not show up on fing after i tried to add it to my network

skipfire, 2025-01-17:
Ok, the error you got sounds like the network management tool may have been changed away from wpa_supplicant, possibly to Network Manager.  In putty can you try the command `nmcli conn` and if that says not found try `sudo nmcli conn`?  It's got an easy mechanism for making hotspots and is what the newest Pi OS uses as well.

grsthegreat, 2025-01-17:
i tried both, they each say
Command not found

skipfire, 2025-01-17:
I don't know if it's that the hotspot is blocking wpa_supplicant from starting, if it's been replaced by something and if so what, or if there's something else going on.  Unfortunately not knowing how it is setup I'm not sure what else to suggest.  You could rebuild the OS yourself, but you may lose any of the hardware & OS configs that were custom with the unit you bought.

grsthegreat, 2025-01-17:
ya, the problem is i know nothing about rebuilding pi devices. i guess im screwed

grsthegreat, 2025-01-17:
all i find is colts24-7 does that sound familiar?

skipfire, 2025-01-17:
No, and Google only seems to return the Indy Colts.

grsthegreat, 2025-01-17:
ok, i entered my genmon unit by typing httpL//wifisetup.pi8000? and i see my unit now. 

i see submit register
submit logs
backup
log files

which should i copy or download

skipfire, 2025-01-17:
Backup should give you a zip file download and that has the genmon configs

grsthegreat, 2025-01-17:
as im at a total loss, ill try contacting you directly. maybe i could send my device and enclosure to you and you can update it to newest device. ill try contacting you thru marketplace. i cant seem to see a link to pintsize here

skipfire, 2025-01-17:
@grsthegreat https://pintsize.me or https://www.facebook.com/profile.php?id=61555703325298

grsthegreat, 2025-01-17:
i sent you message on marketplace. thanks