jgyates/genmon #1245: genmon webUI won't start without generator connected

noumenon272, 2025-02-01:
I'm trying to get ready for my generator install by getting genmon setup ahead of time, but I can't get the web UI to load. I plan to connect over USB. I am using a fresh Raspbian install on a 3b+ and have run the install script without success. 

the logs only show that it can't find the generator, so I'm assuming things are breaking b/c its not plugged in. is this expected behavior? it would be nice to be able to set it up ahead of time.

`pi@genmon:~ $ cat /var/log/genmon.log 
2025-02-01 06:45:26,937 : New version detected: Old = V1.19.05, New = V1.19.05
2025-02-01 06:45:26,946 : Selected Controller: generac_evo_nexus
2025-02-01 06:45:26,961 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:330
2025-02-01 06:45:26,961 : Possible cause is invalid serial port specified.
2025-02-01 06:50:21,523 : Selected Controller: generac_evo_nexus
2025-02-01 06:50:21,540 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:330
2025-02-01 06:50:21,540 : Possible cause is invalid serial port specified.
`

thanks

skipfire, 2025-02-01:
GenMon does require that the port exist (unless using serial over TCP).  You do not have to plug into the generator to continue setting up, but you do have to have your serial to USB device plugged in even if there is no device on the other end.

noumenon272, 2025-02-01:
that worked!

thanks

closed 2025-02-01 by noumenon272