jgyates/genmon #1246: add device field to pushover integration

noumenon272, 2025-02-01:
I’d like to be able to specify which devices pushover sends messages to. and ideally, which devices get which messages. My wife is going to get annoyed with me if she gets 5 pushover messages every time I mess with genmon. She doesn’t need to know when the software needs updated, however, I would like for her to get power notifications. 

thanks

jgyates, 2025-02-01:
I will add this to the list, but in the interim you may want to use email as this functionality already exists for email. You can define multiple emails address, each with a different set of categories.

jgyates, 2025-02-14:
I am going to close this thread. I will update this thread once the feature has been added to the code.

closed 2025-02-14 by jgyates