jgyates/genmon #1248: Not switching power before shutting down

m0ngr31, 2025-02-04:
### Steps to Reproduce

My house is running off-grid. I have a solar setup with batteries. My inverter will start the generator automatically once the voltage gets too low and it'll run for like 3 hours to top it off.

Through Home Assistant, I'm using MQTT to run the generator for a fixed amount of time (say 30-60 minutes in the night depending on my batteries voltage). This usually provides enough to last until morning when the sun comes up in the winter - so my generator runs a lot less than the inverter would.

I'm having an issue with how the generator starts/stops through MQTT - I'm using `setremote=starttransfer`, and `setremote=stop` to start and stop it. It does seem to work properly, but it doesn't actually stop the transfer of power until it turns off totally if that makes sense...

It'll receive the stop command, but the transfer switch doesn't go until the generator stops ~1 minute later. The issue with this is that all the devices in my house will flicker: TVs/monitors will lose power and come back, but things like the microwave and stove clock are not effected. I have a UPS for my computer and router, but it's a huge pain if anyone is watching a show.

The inverter starting/stopping the generator doesn't have this problem.

### Expected Result

Switch the transfer switch before the generator turns itself off

### Actual Result

[genmon_logs.tar.gz](https://github.com/user-attachments/files/18661656/genmon_logs.tar.gz)

### Version Info

v1.19.05 (was previously on v1.18.* and had the same result)

m0ngr31, 2025-02-04:
Thanks for the quick reply :)

> I am assuming you are using two wire mode in your off grid setup.
Yes

> In an on-grid situation, when stoping after a start/transfer command is issued, the controller will dis-engage the transfer switch (back to the utility) first then enter a cool down period where the engine will continue to run for a few moments while not connected to a load. I do not know the order of things for an off-grid setup as this is firmware controlled.

It should be setup in this way I think - the "utility" power is just coming from my inverter instead of the grid. Unless I understand it incorrectly...

> Also, when dis-engaging the transfer switch to move back to utility power it is normal for lights to flicker as the frequency of the generator power is not in phase (sync) with the utility power. It could be that this is what is happening with your solar battery sourced power? Some equipment is more sensitive to changing the phase of the AC power.

With the 2 wire setup I don't get the flicker is the only thing I have to refute that. So I'm not sure what to do. Sometimes I get lights flicker, but never TV - so maybe it's just more of an issue comparing them to each other?

m0ngr31, 2025-02-12:
After sitting down and playing with it for a little bit tonight, the 2 wire setup is behaving the exact same way as Genmon (through MQTT and web interface) - neither of them will trip the switch back to my batteries until the generator shuts off. Maybe it's an issue with the transfer switch or my inverter?

Is there a way to manually flip the switch from Genmon? Curious if there is something else I can experiment with

skipfire, 2025-02-12:
What is the order of your equipment?  I believe the ATS doesn't flip before shutdown unless there is power on the input side.  Could you have the batteries send power to the generator so it knows there is alternate power available and will flip off the ATS then let the generator run a cooldown?

jgyates, 2025-02-12:
It is my understand that the controller only will set relay to an active state and this is the only mechanism that will signal the transfer switch from the controller. I would check that this signal is activated or inactive. The name and pin number of the signal  should be listed in the install guide for your generator but someone else here may know that off the top of their head. If this signal is in the inactive state then I would guess that is not switching back due to the safety mechanism built into the ATS. If the signal is in the active state then this is likely a limit of the controller firmware.

m0ngr31, 2025-02-12:
> I believe the ATS doesn't flip before shutdown unless there is power on the input side.

That's probably correct

> Could you have the batteries send power to the generator so it knows there is alternate power available and will flip off the ATS then let the generator run a cooldown?

That's a good question. I don't know that the firmware my inverter is running would allow that. I'd have to do some research

jgyates, 2025-02-14:
i am going to close this issue but feel free to post any follow up to this thread.

closed 2025-02-14 by jgyates