jgyates/genmon #1251: Genmon reports Utility "Failure" when we are not observing utility power

twistedroutes, 2025-02-13:
### Steps to Reproduce

Configure genmon (/etc/genmon/genmon.conf):
`disableoutagecheck = true`

Run genmon with no utility power

Observe Status page:
Status :
    Switch State : Automatic
    Engine State : Stopped
    Utility Power : Failure

### Expected Result

With disableoutagecheck=true I expect Utility Power to be "N/A" vs "Failure" (ore even "healthy")

Observe Status page:
Status :
    Switch State : Automatic
    Engine State : Stopped
    Utility Power : N/A

or

Observe Status page:
Status :
    Switch State : Automatic
    Engine State : Stopped
    Utility Power : Healthy

For my briggs and stratton this is coming straight from the register per the config file:
```
"status": [
   ...
       {
      "reg": "004f",
      "reg_type": "input",
      "mask": "4000",
      "value": "0000",
      "type": "bits",
      "title": "Utility Power",
      "text": "Failure"
    },
    {
      "reg": "004f",
      "reg_type": "input",
      "mask": "4000",
      "value": "4000",
      "type": "bits",
      "title": "Utility Power",
      "text": "Healthy"
    },
    ...
]
```
Would like to be able to ignore the Failure when disableoutagecheck=true (in my case there is no utility as I'm off-grid)... maybe a new option "outagestateifignored": "N/A"

### Actual Result

Observe Status page:
Status :
    Switch State : Automatic
    Engine State : Stopped
    Utility Power : Failure

### Version Info

1.19.05

jgyates, 2025-02-13:
The disableoutagecheck setting will disable the email and short message notifications for an outage, disable the outage log and the display of the outage page on the web interface. This disables the thread that polls for an outage in genmon. It does not change the display of the controller status. What you are asking it to stop the bit in register 004f from being displayed. 

I would suggest one of three options:
1) Ignore this text on the web interface as it is just informational representation of the controller status
2) I could change the word "Failure" to "Unhealthy" in the Briggs and Stratton config file. 

From the perspective of genmon this is just a bit that the controller sets. No action is taken if you have disableoutagecheck set to true. The Information I have on this bit labels it as  either "Healthy" or "Unhealthy".

twistedroutes, 2025-02-13:
Understood. You mentioned a 3rd option?  Note that this also appears in the
emails and I keep opening the email and seeing failure (trained to lol)

My thinking here though is, if we're saying "ignore the utility" then it
would be good to be able to alter the value *represented* in the config
file.  In this manner, the same config file can be used for someone that
has utility and those that don't.

Or even be able to set an alternative value in the config file based on a
config variable:
{
      "reg": "004f",
      "reg_type": "input",
      "mask": "4000",
      "value": "0000",
      "type": "bits",
      "title": "Utility Power",
      "text": "Failure",
      "text_check_var": "disableoutagecheck",
      "text_check_alt": "N/A"
    },

In this way, before we present text, we see if text_check_var is defined,
if it is then check if the identified var is defined AND true....
just a thought.  I'd rather not alter the code or config beyond what
everyone else uses but contribute to modifications that benefit everyone
(and avoid overwriting my own config changes lol)...

jgyates, 2025-02-13:
Sorry, I was thinking about a third option but decided that it was the same as the 2nd option. 

The status emails just display the info the web interface does. There is a separate email that only contains outage info if you have the outage check enabled. In short, there is come code that continuously checks for outages, if one is detected the log is updated and a notice is sent. The disableoutagecheck option disables this functionality. 

Regarding the "text_check_var", I can add this suggestion to the feature list.

twistedroutes, 2025-02-13:
Thanks so much for your insights and help...oh and code...

while you are thinking about the var feature, I'm starting to work on
https://github.com/jgyates/genmon/discussions/1249
For now it will be a relay that I operate externally to genmon, but I'd
like to add a button to `maintenance` to handle it.  Unfortunately right
now, all buttons defined in the conf files (via onewordcommand) must have a
"reg" value.  What I'd like to do is to have a button that either:
 - sets a gpio pin when pressed (in my case to trigger the generator reset
as the GPIO would trigger the relay to the OPEN state)
 - OR
 - runs another python script when pressed (which would then handle the
setting of the GPIO pin which would open the relay etc.)

twistedroutes, 2025-02-13:
I meant to include:

{
"onewordcommand": "powerswitchreset",
"title": "Reset Generator Power via Relay",
"command_sequence": [
{
"comment": "set a GPIO Pin that a relay trigger is connected to",
"action": "gpio",
"action_param": [17, true]
}
]
},

jgyates, 2025-02-13:
I will think on this as well. I would make it a separate json file that can be read in by any controller if optionally enabled. I would also need to make sure the raspberry pi stuff is sufficiently encapsulated and validated since there are more than a few people running in docker images or on non pi systems. To date I have only had pi specific things in add on programs, not genmon proper.

jgyates, 2025-02-13:
I updated the feature request list: https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will update this thread when it is compelte.

jgyates, 2025-02-14:
I added this feature. 

The disableoutagecheck is not available on the Settings page (as it was for Evolution / Nexus controllers) and for Briggs and Stratton it will remove the Utility information from the display if disableoutagecheck is enabled.