jgyates/genmon #1255: Custom gauge not working since last upgrade.

T-C123, 2025-02-21:
### Steps to Reproduce

Must have add-on installed.

### Expected Result

Custom oil pressure gauge using external sensor, circuitry and MQTT Sensor Input Add-On reports oil pressure in text and gauge format.

### Actual Result

Custom gauge stopped displaying after last Genmon update, text data still reported in "External Sensors" on Genmon "Status" screen.

### Version Info

Genmon v. 1.19.05

Debian v. 11.11

### Logs

Generac 18 kW
Logs and registers submitted.

jgyates, 2025-02-21:
@T-C123 

I got your logs. 

It it hard to say for sure without your genmqttin.conf data. I did see one error in genmon.log:

Error in SetupCommonTiles: sensor_bounds: 'min' : controller.py:2167

I added a line that would allow this to be zero if the minimum is not set. Try updating from the About page and let me know if that works or not. The version number has not changed. If that does not work then send the the contents of /etc/genmon/genmqttin.conf

T-C123, 2025-02-21:
The new update fixed it, Thanks! I couldn't attach the genmqttin.conf file so I pasted it below. Thanks for your help.

Tom

[genmqttin]

# NOTE: single entries for topics, labels, types, units, nominal_values and 
# maximum_value do not require comma separators

# Required. a list of comma separated MQTT topics to subscribe. This add on will 
# monitor each topic and import the data into genmon for display. The maximum 
# number of topics to monitor is four
topics = Oil pressure

# Required. A list of labels that corresponded to the list of topics. These labels
# will be used to display the data read from the MQTT broker in the User Interface 
# of genmon. Each label must be unique and the number of labels must match 
# the number of topics.
labels = Oil Pressure

# Required. A list of comma separated units. These correspond with the list of 
# topics. If a unit is not used then a blank will be used to keep alignment 
# with the number of topics. Then number of units (including blank units) must
# match the number of topics.
units = PSI

# Required. A list of comma separated types. These correspond with the list of 
# topics. Then number of types must match the number of topics. Valid types are:
#
#   fuel - Used for monitoring fuel gauges. The data is a percentage 0 - 100 of 
#       the fuel tank level. Max should be set to 100. Since this is a 
#       percentage units can be none or %.
#   temperature - Used for temperature sensors. Nominal and Maximum values are 
#       sensor specific
#   power - Used for importing power readings into genmon. This  should be used 
#       for reporting the total power (sum of all legs) of the system. Units 
#       should be W or kW.
#   current - Used for importing current readings into genmon. This should be 
#       used for reporting the total current (sum of all legs) of the system. 
#       Units should be A.
#   voltage - Used for reporting voltage. This is the total voltage for the 
#       system.
#   voltageleg1 - Used for reporting a single voltage leg. This value will be 
#       added to voltageleg2 for the total voltage. Units should be V.
#   voltageleg2 - Used for reporting a single voltage leg. This value will be 
#       added to voltageleg1 for the total voltage. Units should be V.
#   pressure - This is for reporting a sensor pressure. 
#   ct1 - This is used for report the current for the first leg in a multi-leg 
#       system. Units should be A.
#   ct2 - This is used for report the current for the second leg in a multi-leg 
#       system. Units should be A.
#   ctpower1 - This is used for report the power for the first leg in a multi-leg 
#       system. Units should be W or kW.
#   ctpower2 - This is used for report the power for the first leg in a multi-leg 
#       system. Units should be W or kW.
types = pressure

# Required. The nonomal value helps the genmon UI determine how to display the 
# gauge colors. This is a list of comma separated units. These correspond with 
# the list of topics. 
nominal_values = 30

# Required. The maximum value helps the genmon UI determine how to display the 
# gauge colors. This is a list of comma separated units. These correspond with 
# the list of topics. 
maximum_values = 50

# Required. Address of MQTT Broker server. This value should be changed to match 
# the IP address of your MQTT server.
mqtt_address = 127.0.0.1

# Optional. This value is used if your MQTT server requires authentication.
# This is the username.
# NOTE: a return code of 5 when connecting to the MQTT server typically means
# that the username / password was not provided or is incorrect.
username = generac430@gmail.com

# Optional. This value is used if your MQTT server requires authentication.
# This is the password.
password = ygiz zdcz wmjq cutv

# Optional. This value is the IP address of the system running the generator
# monitor software. If the genmqttin.py program is running on the same system as
# genmon, then this parameter is not needed.
monitor_address =

# Optional. This parameter is the port of the MQTT server in a decimal number.
# If this value is not present then 1883 the default.
mqtt_port = 1883

# Optional. This value will be added to the beginning of the MQTT path. The
# default path used is 'generator/' however you could add set root_topic=Home
# to make the path be Home/generator.
root_topic =

# Optional. Full path to Certificate Authority file. Leave empty to not
# use SSL/TLS. If used the port will be forced to 8883.
cert_authority_path =

# Optional. Full path to the client certificate file. Leave empty to not use MTLS
client_cert_path = 

# Optional. Full path to the client key file. Leave empty to not use MTLS
client_key_path =

# Optional. TLS version (Integer). Specify the TLS version used. Default is 1.
# Must be 1.0, 1.1, or 1.2. This is ignored when a CA cert file is not used.
tls_version = 1.0

# Optional. Defines the certificate requirements that the client imposes
# on the broker. Used if Certificate Authority file is used. None,Optional,
# and Required are the valid options. Required is the default.
cert_reqs = None

# unique identifier for each instance of genmon running. This can be any value
# if you are running one instance of genmon. Otherwise each instance must be
# unique
client_id = sensor
use_for_power_fuel = true

skipfire, 2025-02-21:
@T-C123 you may want to delete that content and then change your password, it has email and password in plain text.

jgyates, 2025-02-21:
@T-C123 

I removed those lines from the thread but I would still change your password. I did not realize you had authentication to your MQTT broker.

closed 2025-02-21 by jgyates

T-C123, 2025-02-21:
Thanks guys! I was in a big hurry and just pasted the text without reading it.