jgyates/genmon #1264: Engine RPM SNMP

msteil4vse, 2025-03-24:
### Steps to Reproduce

Hello,

I need someone to verify this observation.  I monitor the Genmon Pi using SNMP to log valorous conditions in our network monitoring system.   We had a generator that that the RPM was surging from time to time, and as it was at a remote location we were attempting to log and graph the engine RPM via SNMP.  During the engine difficulty the web interface showed the RPM variation, and the SNMP RPM query was all over the board ranging from 0 to 3600 RPM.   Now, after the engine issue has been repaired, I am observing the engine RPM from the web interface the RPM is stable running at 3600 RPM as it should be, however if you look at the attached graph you will see an oscillation from 600 RPM to 3600 RPM back and forth.   I have done some SNMP queries directly from my desktop to verify the MIB 1.3.6.1.4.1.58399.1.0.0.7 and that query is constant with what the network monitoring system is graphing, I get a return of value of either 36XX or 6XX RPM. 

[image]

### Expected Result

I'd like someone to verify that see can duplicate the same issue before submitting this as a "bug".  A side issue I also would be interested to know how many people are using SNMP for monitoring. 

### Actual Result

See above 

### Version Info

Version V1.19.05

### Logs

[genmon_logs.tar.gz](https://github.com/user-attachments/files/19433982/genmon_logs.tar.gz)

jgyates, 2025-03-24:
Hi @msteil4vse 

What controller do you have? Is it Evolution 2 or 1 or something else?

You logs don't show the communication stats. Can you post that info from the Monitor page, or if outbound email is setup you can submit your logs and that info will be included.

msteil4vse, 2025-03-24:
I knew I was forgetting something.  The controller is an Evolution 2.0. 

 

 I did submit the logs and registers.  I will submit them again momentarily.  Advise if you get them

 

 

Monitor :

 

    Generator Monitor Stats :

        Monitor Health : OK

 

        Controller : Evolution 2.0, Air Cooled

 

        Run time : Generator Monitor running for 13 days, 19:42:04.

 

        Power log file size : 0.01 MB of 15.00 MB

 

        Generator Monitor Version : V1.19.05

 

        Update Available : No

 

 

 

    Communication Stats :

        Packet Count : M: 38284941, S: 38284935

 

        CRC Errors : 0

 

        CRC Percent Errors : 0.00%

 

        Timeout Errors : 5

 

        Timeout Percent Errors : 0.00%

 

        Modbus Exceptions : 2428

 

        Validation Errors : 0

 

        Sync Errors : 0

 

        Invalid Data : 0

 

        Discarded Bytes : 0

 

        Comm Restarts : 0

 

        Packets Per Second : 64.12

 

        Average Transaction Time : 0.0308 sec

 

        Modbus Transport : Serial

 

        Serial Data Rate : 9600

 

 

 

    Platform Stats :

        CPU Temperature : 115.16 F

 

        Pi Model : Raspberry Pi 3 Model B Plus Rev 1.3

 

        Pi CPU Frequency Throttling : OK

 

        Pi ARM Frequency Cap : OK

 

        Pi Undervoltage : OK

 

        CPU Utilization : 5.45%

 

        OS Name : Raspbian GNU/Linux

 

        OS Version : 11 (bullseye)

 

        System Uptime : 16 days, 1:40:43

 

        Network Interface Used : eth0

 

        System Time : Monday March 24, 2025 15:45:21

 

 

 

    Weather :

        Current Temperature : 49.51 F

 

        Conditions : Overcast Clouds

jgyates, 2025-03-24:
Hi @msteil4vse 

I don't use SNMP to monitor RPM, but I do use it for monitoring fuel level. After looking at you log files my first guess would be that your SNMP community string is not matching between all of the calls. You log file located at /var/log/gensnmp.log has the error "Invalid community string" a lot. The log file will show the expected and the configured community string. If this error occurs the results returned are not complete as the request is stopped immediately.

Let me know if that solves your issue of it the problem still persist.

msteil4vse, 2025-03-25:
Well, this is helpful for that aspect.   I did a packet capture on the Ethernet port and did not see the SNMP packets when entries were found in the log.  I then did a ipconfig and did not realize the WLAN was connected to anything.  Once found that, it was easy to see the errant SNMP requests were coming in on the WLAN.  I got that shut down and that takes care of that red herring.

I ran the generator after getting the community string issue resolved, and as you can see below, the graph still indicates the oscillation.  As I did previously I did  an snmpget repeatedly from a terminal not associated with out network monitoring system that I had remote access to on that network and you can see the direct query indicates the same oscillation that is seen in the graph.

I have submitted the logs and registers again, for your review .

[image]

[image]

jgyates, 2025-03-25:
I have tried duplicating this on my generator with snmpget and have not been able to repeat this behavior. 

One thing we can try would be to monitor the RPM without the SNMP add on. To do this go to the live register view page in the web interface by clicking the icon in the upper right once. This will take you to the live register view page. This page will show the contents of the registers as they change as long as the web interface page is active. If you close the browser then logging will stop. You can locate the RPM register on the page and if you hold your mouse over the register data it will show a graph of the historical data. It should look like this:

[image]

Let me know if you see the issue in this view.

Also, are you monitoring multiple generators? If so, do you see this in another instance? 

How frequently are you polling via SNMP?

What are the values of polling interval, numerics only and force integer in the genmon SNMP add on?

msteil4vse, 2025-03-25:
I did Part one, and you will see by the attached movie, that the RPM in the register is stable, but the graphed output from SNMP is not consistent with what the register is showing.

I do monitor other generators, however never before have tried to monitor the RPM via SNMP, and actually the only reason I started on this particular generator is  I was not aware of the register feature, and we were trying to determine what was happening on this generator with it had problems so I stated poling for output cycles and ending RPM..  Now that it is fixed it is kind of an academic exercise.   I can set it up on a different generator if you want me to, but it will take some time to set it up and test, which might take a day or so,  with my other workflow.

The polling is set to once a second at the moment, I did try backing it off to once every 5 seconds but same result, just longer windows between oscillations.  -- I have looked at the load average on the Pi, and it seems to peak out at .4 wit with the  1 sec interval and all the other SNMP data I am gathering  

Last item "What are the values of polling interval, numerics only and force integer in the genmon SNMP add on?" lost something in translation, not certain what you are asking for here.

[image]

msteil4vse, 2025-03-25:
The movie did not attach, and I had trouble getting a screen shot without the graph disappearing, so the image below is a capture of the final moments of the window capture movie.

[image]

msteil4vse, 2025-03-25:
Ok, now I understand

[image]

jgyates, 2025-03-25:
also, when you get a moment can you send your logs while the generator is running. The email includes a snapshot of the registers so I want to see if there is another register that has a value of around 600-602 as this corrosponds to the low value of the RPM you are reading. Just a shot in the dark, but it may help.

jgyates, 2025-03-25:
I tried my test again with "Numerics only" enable like you have and I still could not duplicate. Let me know if you see the results on another generator when you get a chance to test.

jgyates, 2025-03-25:
There is another thing we could try:

edit the file /etc/genmon/gensnmp.conf and add the line:

     debug=True

save the file, then restart genmon. you can restart genmon on the ssh console from the genmon directory with this command:

     ./startgenmon.sh restart

Once you restart, this will add additional data to the /var/log/gensnmp.log file. Read the SNMP data until you get a bad result, then submit your logs from the About page. Try to limit the time between the error and when you submit the logs as the log files  will rotate and archive if they get too large. Once you have submitted the logs then edit the /etc/genmon/gensnmp.conf file and set

    debug=False

Then restart genmon as described above. In general you don't want to leave the debug option enable for long periods of time since over time it will shorten the life of your SD card with the extra writes.

Hopefuly this test will show when the SNMP add on is updated by genmon and when the data is requested from an external source.

msteil4vse, 2025-03-25:
Here is Part 1, I started Generator and submitted logs while Generator was running.  I don't know if this muddies the water any but I am also graphing the Generator Frequency as well.  the images attached are for the same time fram and yuo can see the engine RPM as compared to the Frequency.

[image]
[image]

msteil4vse, 2025-03-25:
#2 -- I did as you asked regarding the debugging info, and I immediately get back:

2025-03-25 12:45:25,357 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,059 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,062 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,452 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,456 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,459 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,463 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:26,693 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:27,376 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:27,805 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:28,363 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,055 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,058 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,318 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,322 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,325 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,329 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,450 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,460 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,463 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,467 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,694 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:29,794 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:30,380 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:30,808 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:31,356 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:32,060 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:32,064 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:32,321 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:32,324 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-25 12:45:32,328 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608

I did not run the generator at this point.

jgyates, 2025-03-25:
you may be looking at a failing SD card. 

The code on this line has nothing to do with the debug code. You can try reloading genmon without but I have seen a failing SD card do strange things. The SNMP support uses a [python library called pysnmp](https://pypi.org/project/pysnmp/) to implement the SNMP stack. A SD card intermittency could be causing the failures you are seeing if it is causing issues with the SNMP python module.

Here is the line of code that is causing the error in your logs:

https://github.com/jgyates/genmon/blob/9ad9c2309133fe289811141a7a79fe11183c1968/addon/gensnmp.py#L608

All the debug=True does is allow the function self.LogDebug to write to the log file.

You may try rebooting your pi. 

The test on another system will confirm the SD card theory.

msteil4vse, 2025-03-25:
Ok, I can test another generator installation, but I get pulled in too many different directions during the work day to concentrate on setting the monitor up on a different generator, and put fires out at the same time.  So give me 24 hours or so and I will advise.

jgyates, 2025-03-25:
No rush, I understand completely.

msteil4vse, 2025-03-26:
Just a quick update, I did get the RPM graphing via SNMP configured on a second Generator, and it is exhibiting the same characteristics.  FYI, this is the one that previously we conversed with Genmon and difficulties after OS upgrade, and the relevant aspect being the SD card in this unit is only a month or two old, and when deployed it was a new SanDisk Extreme Pro card.  When you have time, can you make a list of the data you'd like collected on second generator?

msteil4vse, 2025-03-26:
forgot to post graph...

[image]

jgyates, 2025-03-26:
I loaded a test system with my emulator running Evolution 2.0 registers and I am still not able to reproduce the issue. If you can perform the test with debug enabled as described above and send the logs that would give me more info.

Is there anything else unique about this installation? for example is it running in a VM, or have any project files been modified, etc?

jgyates, 2025-03-26:
also, you may try testing with no other add ons running. From your logs it appears you are running a few add ons. I don't know if an issue that would cause this with multiple add on program, but I am just trying to get this down to the least number of variables so I can duplicate.

msteil4vse, 2025-03-27:
There has to be some aspect of the story that we are missing.  I have a new observation, while I was trying to gather the requested information, this time on the second generator configuration, I enabled SNMP debug, and after restarting Genmon I get a stream of the following in the log:

2025-03-27 07:57:52,901 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608
2025-03-27 07:57:52,905 : Error in SnmpCallbackFunction: 'list' object has no attribute 'strip' : gensnmp.py:608

Obviously we have seen this before, so I initially ignored the entries and attempted to do some SNMP queries and I found that I could not get a reply to snmp querys when I did a simple snmpwalk.  (snmpwalk -v2c -c monitor 192.168.30.31 1.3.6.1.4.1.58399.1.1.0)  

If I disable the debug mode then the SNMP queries work normally. 

I have rebooted the Pi, no effect

I have turned off all add-ons same result.  System is now running debug off, and only add-ons active are SNMP and Mopeka.

Lastly, if it would be more efficient, and you are interested, contact me privately, and we can make some arrangements to allow you to see in real time and allow you to make some direct observations.

jgyates, 2025-03-28:
I have had the debug flag on in all of my testing. Let's check the library version you are using. What is the output of this command:

sudo pip freeze

msteil4vse, 2025-03-28:
Attached is the output of both installations. I briefly did a "diff" and the output of that showed a bunch of differences, so hopefully this will be helpful.

[Headend.pip.freeze.txt](https://github.com/user-attachments/files/19506218/Headend.pip.freeze.txt)
[House.pip.freeze.txt](https://github.com/user-attachments/files/19506219/House.pip.freeze.txt)

jgyates, 2025-03-28:
without posting any passwords, etc, can you post the contents of your /etc/genmon/gensnmp.conf file.

The error in your log for "'list' object has no attribute 'strip' : gensnmp.py:608" looks like the file is parsing the community string as a list instead of a string. How are you editing this file when you add the debug line?

jgyates, 2025-03-28:
also, what other software is loaded on the pi? There are a lot of python libraries installed in your pip output that are not on my system.

[freeze.txt](https://github.com/user-attachments/files/19506561/freeze.txt)

msteil4vse, 2025-03-28:
Ok, I think I found and ID10T error.   I just pasted "debug=true" at the top of the config, and totally missed the entry lower in the file setting debug = False.  I am retesting now.

As for editors/editing files you may not understand this one depending on your age, someone lurking here will that understands will get a chuckle and have some understanding.  I use Joe's on Editor for most everything, and nano when I have to.  Back in the day, when there was no alternative I used vi, and detested every moment using it.   Being child of the "WordStar" days I know all the command set reflexively, so Joe being a WordStar clone that's what I gravitate towards.  does that date me or what?  ;-) 

More to come shortly. 

[gensnmp]
#debug=true

# number of seconds between polls for generator data.
poll_frequency = 2

# community name
community = XXXXXXX

# include more info in gensnmp.log for debugging purposes.
# default to False to save on the SD card life
debug =  False

# This is the genmon enterprise ID.
enterpriseid = 58399

#optional override of default snmp port to use
snmpport = 161
use_numeric = true
use_integer = false

jgyates, 2025-03-31:
I got your latest logs and I think this allows me to see the issue.

The add on parses the data, looking for key words. In this case it looks for "Status/Engine" and  "RPM". 
In normal cases the RPM would be reported under "home/Status/Engine/RPM "

The problem is that, when you also enable "Display Unknown Sensors" on the setting page, genmon will output an additional value under "home/Status/Engine/Unsupported Sensors/Raw RPM Sensor"

This causes the Add On to dump both the data from the Raw RPM sensor and the normal RPM data to the same OID. This would also effect the MQTT add on. 

To fix the issue I renamed the "Raw RPM Sensor" to "Raw Revolution Sensor". 

If you update from the About page this should fix the problem. The version number has not been updated. Let me know how your testing goes.

msteil4vse, 2025-03-31:
I applied the update and it works flawlessly..   Thank you for your efforts and sorry it took so much to drill down to the problem. 

-- Re not finding this on your system, I'm probably the only one that would think of logging and graphing engine RPM..  :-)

[image]

jgyates, 2025-03-31:
No worries, I am glad it is working as expected now. I actually log and graph my RPM, I just don't run the "Display Unknown Sensors" on my production system (the one that does the logging and graphing) and did not think to turn it on on my test system until I saw your debug logs.

msteil4vse, 2025-04-01:
Good AM, the update I did above was on a somewhat older install on a 32 bit installation. this AM I attempted to update the other unit we were testing on which is a 64 bit installation, and post upgrade I see in the genmon log

2025-03-31 19:57:10,702 : GenMon Loaded for site: HeadEnd GenMon using python 3.11: VEnv: True
2025-03-31 20:21:12,284 : Generator Monitor Shutdown
2025-03-31 20:21:18,116 : Selected Controller: generac_evo_nexus
2025-03-31 20:21:20,025 : GenMon Loaded for site: HeadEnd GenMon using python 3.11: VEnv: True
2025-03-31 20:23:15,131 : Selected Controller: generac_evo_nexus
2025-03-31 20:23:15,150 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:329
2025-03-31 20:23:15,150 : Possible cause is invalid serial port specified.
2025-04-01 00:10:56,850 : Selected Controller: generac_evo_nexus
2025-04-01 00:10:56,870 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:329
2025-04-01 00:10:56,871 : Possible cause is invalid serial port specified.

Commands I issued:
apt -y update
apt dist-upgrade
/home/genmonpi/genmon/genmonmaint.sh -u
/home/genmonpi/genmon/startgenmon.sh restart

Do you have any sage advice on how to correct?  otherwise I will dig into it later today. 

Mark

jgyates, 2025-04-01:
Based on the log output I would check the serial port and make sure you settings match the ports present. You can type this command from the console:

    ls -la /dev/ser* 

This will display the onboard serial ports. Or you can type

    ls -la /dev/ttyUSB*

to show USB based serial ports. 

Make sure this matches the serial port in the genmon settings.

The file /var/log/myserial.log may have more info.

msteil4vse, 2025-04-01:
looks like the update must have removed /dev/serial0  and seems to have totally fouled up the serial interface.   I am trying to just build a new SD card hopefully a faster solution than debugging the current card..  Genmon won't run, when you make your backups from the Web interface, are you just making a tarball of /etc/genmon?  More directly what all are you backing up ?

jgyates, 2025-04-01:
The genmon software (and the update) does make any changes to the system serial ports configuration. Genmon only reads and writes to the serial port to send data. Only the install script modifies the serial configuration via the raspberry pi boot files. The modification of the boot files only occurs in serialconfig.py and that is only called from the install script (./genmonmaint.sh -i)

The backup just makes a tar.gz of conf files, outage log, and power log, etc. The bash script that does this is located here:

https://github.com/jgyates/genmon/blob/85f87493dc7dbff70aede23a83527635eadd631a/genmonmaint.sh#L340

skipfire, 2025-04-01:
I heard a report over the weekend of a user with a similar issue, after a dist-upgrade the symlink for the serial port was removed.

jgyates, 2025-04-01:
Good to know, thanks

msteil4vse, 2025-04-01:
For the record I experimentally installed a fresh install of Pi OS with the Pi imager  installing 64 bit  release 2024-11-19 that was cached locally.  Then did the following:

sudo apt update
sudo apt -y install joe git ftp
sudo apt -y full-upgrade
git clone https://github.com/jgyates/genmon.git
cd genmon/
./genmonmaint.sh -i
cd ../
ftp -i 192.168.XXX.XXX
tar -zxvf Headend_genmon_backup.tar.gz 
sudo cp -v ./genmon_backup/* /etc/genmon/
cd genmon/OtherApps/
sudo python3 serialconfig.py -e -b
sudo reboot

cat /var/log/genmon.log

2025-04-01 11:28:45,400 : Selected Controller: generac_evo_nexus
2025-04-01 11:28:45,415 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:329
2025-04-01 11:28:45,416 : Possible cause is invalid serial port specified.

Genmon fails to start/run  

Next step in progress is to do same procedure and skipping updates.  Will advise.

msteil4vse, 2025-04-01:
Update, repeated previously documented steps  omitting "sudo apt -y full-upgrade".  The installation works normally.

jgyates, 2025-04-01:
what is the contents of the file /var/log/myserial.log

msteil4vse, 2025-04-01:
2025-03-31 20:23:15,146 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-03-31 20:23:15,147 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-03-31 20:23:15,147 : Error on serial port init!
2025-04-01 00:10:56,866 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-01 00:10:56,867 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-01 00:10:56,868 : Error on serial port init!
2025-04-01 10:21:14,560 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-01 10:21:14,563 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-01 10:21:14,563 : Error on serial port init!

Previously I did try recreating the link serial0 -> ttyAMA0, and then I ran 

python3 /home/genmonpi/genmon/OtherApps/serialtest.py

returning 

Using python serial library V3.5
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes 

This is the point I got disinterested in debugging and decided to make a new SD card.

msteil4vse, 2025-04-01:
I did install new SD card in generator installation, and I have problematic card in my office installed in spare Pi, should we need further info.

skipfire, 2025-04-02:
The loopback failure is frequently caused by either not having the right 2 pins connected, or genmon being live when you try to do the loopback as serial is an exclusive lock.

@jgyates I did some testing and found that `sudo apt update && sudo apt upgrade -y` wiped my symlink for serial0.  I don't know what specific update caused it, but it was not a dist-upgrade call.  Might be something that starts seeing more issues soon.

jgyates, 2025-04-03:
Did you reboot after the update/upgrade?

The project program ./genmon/OtherApps/serialconfig.py modfies the /boot/config.txt file and makes a copy of the original named /boot/config.txt.bak1. I would compare the current (post upgrade) /boot/config.txt with the backup copy of the original (the bak1 is the config.txt pre update/upgrade). My guess is that it is overwriting or replacing the config.txt file, but that is a guess.

jgyates, 2025-04-03:
Same issue resolved?

https://github.com/jgyates/genmon/issues/1270

msteil4vse, 2025-04-04:
I finally had a chance to verify this is the same issue, and yes it is.   When I initially discovered link serial0 -> ttyAMA0 was missing I did re-create the link, and reboot, but that did not fix the problem, but after reviewing #1270 I changed the config file to point at  ttyAMA0 and re-boot.  Now it works normally.

closed 2025-04-07 by jgyates