jgyates/genmon #1266: Genmon install missing ability to set switch state

jamesduke, 2025-03-29:
### Steps to Reproduce

Visit Maintenance page
Look for Switch state section

### Expected Result

Ability to set switch state between Auto, On, Off

### Actual Result

Section is missing

[image]

### Version Info

Version V1.19.05
Liquid Cooled Generac, Evolution controller

### Logs

[genserv.log](https://github.com/user-attachments/files/19514851/genserv.log)

[genmon.log](https://github.com/user-attachments/files/19514852/genmon.log)

jamesduke, 2025-03-29:
Prior to a recent software update there was the ability to change this, though it did not appear to work. Has this been removed because it didn't work, or is something wrong with my install

Thanks for the help

jgyates, 2025-03-29:
Hi @jamesduke 

No this has not been remove, but it is only available on system that is it known to work. The Evolution Liquid Cooled 4.5L models this functionality is disabled, for Non 4.5L Liquid cooled models this feature is active. 

I don't remember the exact time that this was determined to not be supported on the 4.5L, it could have been since the last time you upgraded.

You can still issue the command via the command line using the ClientInterface.py program if you have a 4.5L and want to test it out. If it works for you let me know and I can enabled it for specific firmware version for the 4.5L model.

From the genmon folder:

[image]

jamesduke, 2025-03-29:
Thanks for the lightning fast response!

I do have the 4.5L and tried it via the command line - it does **not** cause the controller to switch state. It had been some time since I upgraded following a hardware failure, so that explains why I hadn't seen it before.

closed 2025-03-29 by jamesduke