jgyates/genmon #1270: Genmon not starting after OS upgrade

wrvolz, 2025-04-02:
### Steps to Reproduce

Upgraded OS on 4/1. Today had power outage but could not connect to genmon. RPI was up and could ssh in. Tried restart:

root@Generac:/home/bill/genmon# ./startgenmon.sh  restart
using binary: /home/bill/genmon/genenv/bin/python
Restarting genmon python scripts
Stopping....
Stopping genserv.py
kill: (1660): No such process
Stopping genmon.py
kill: (1638): No such process
Starting....
Starting /home/bill/genmon/genmon.py
Stopping genserv.py
Starting /home/bill/genmon/genserv.py

Notice genserv and genmon not running; Here is part of genmon.log

2025-04-01 01:04:33,052 : Generator Monitor Shutdown
2025-04-01 01:05:51,535 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:330
2025-04-01 01:05:51,537 : Possible cause is invalid serial port specified.
2025-04-02 18:04:46,340 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:330
2025-04-02 18:04:46,341 : Possible cause is invalid serial port specified.
2025-04-02 18:07:19,206 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:330
2025-04-02 18:07:19,206 : Possible cause is invalid serial port specified.

This a RPI 3b+. os-release is:

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

DId a apt-get dist-upgrade shortly after midnight on 4/1. The genmon.log entry at 1:04 is shutting down genmon from the reboot after the upgrade.

It appears something changed in the upgrade.

Thanks,

### Expected Result

Genmon should be running.

### Actual Result

Genmon failed to start after system upgrade.

### Version Info

Genmon is latest version.

### Logs

[genmon.log](https://github.com/user-attachments/files/19576636/genmon.log)

DrTron, 2025-04-02:
Just ran into the exact same issue.
My config file lists "/dev/serial0" as the port to use, but that device doesn't exist.
I use the PiHat.
The port name changed. Do a "dmesg | grep serial" and you'll see that it's now called ttyAMA0.
So change the config file to:
port = /dev/ttyAMA0

wrvolz, 2025-04-02:
That fixed it. Thanks,

Bill Volz
832-724-5183

[91ce0b78-c25b-4026-9bb2-15cb0b8d9a32]

jgyates, 2025-04-03:
/dev/ttyAMA0 is the base name the kernel uses, but the /dev/serial0 is the more linux friendly name. There is a symbolic link for /dev/serial0 that should point to /dev/ttyAMA0. It appears that the symbolic link is gone.  On the pi, you should have a file named /etc/udev/rules.d/10-serial.rules that dictates the symlink rules for serial ports. To reload the rules type this at the console:

     sudo udevadm control --reload-rules
     sudo udevadm trigger

Then type this to validate the link is there:

     ls -la /dev/ser*

Let me know how if this fixes your issue. 

Not sure if this is related but, I upgrade my pis almost weekly with this command: 

     sudo apt-get update && sudo apt-get dist-upgrade

and have not seen the issue on my production or test systems.

DrTron, 2025-04-03:
Interesting. 
I don't have the 10-serial.rules in the udev directory, only a 99-com.rules, which contains this:
```
KERNEL=="ttyAMA[0-9]*|ttyS[0-9]*", PROGRAM="/bin/sh -c '\
        ALIASES=/proc/device-tree/aliases; \
        TTYNODE=$$(readlink /sys/class/tty/%k/device/of_node | sed 's/base/:/' | cut -d: -f2); \
        if [ -e $$ALIASES/console ]; then \
            if [ $$TTYNODE = $$(strings $$ALIASES/console) ]; then \
                echo 0; \
            elif [ -e $$ALIASES/bluetooth ] && [ $$TTYNODE/bluetooth = $$(strings $$ALIASES/bluetooth) ]; then \
                echo 1; \
            else \
                exit 1; \
            fi \
        elif [ $$TTYNODE = $$(strings $$ALIASES/serial0) ]; then \
            echo 0; \
        elif [ $$TTYNODE = $$(strings $$ALIASES/serial1) ]; then \
            echo 1; \
        else \
            exit 1; \
        fi \
```
The `/proc/device-tree/aliases/serial0` does exist, but it looks like this code:
```
elif [ $$TTYNODE = $$(strings $$ALIASES/serial0) ]; then \
            echo 0; \
```
Does exactly nothing. Consequently, udev does not create the alias in /dev/.

My system is Debian 12 on a Zero 2W  (64-bit Raspberry Pi OS).

wrvolz, 2025-04-03:
There is no 10-serial.rules in that directory on my system. There is only a 99-com.rules files. I did the two udev commands and still not /dev/ser*.

I do the same update procedure.

I wonder is there was an error during the upgrade process that I didn't see. I redid the upgrade this morning and nothing was installed so it appears to be all up to date.

Bill Volz
832-724-5183

[a6ee449c-f123-4910-b0c6-8864012f8f53]

DrTron, 2025-04-03:
> /soc/serial@7e201000

Mine does, too. But with the udev-Rule, that never gets called and just goes to "exit 0".

In any case, I see the advantage of using the alias in case AMA0 changes it's name sometime in the future, but right now that's a valid solution.

jgyates, 2025-04-03:
I agree.

closed 2025-04-07 by jgyates