jgyates/genmon #1273: Genmon doesn't seem to be using config settings

ConnorMX25, 2025-04-10:
### Steps to Reproduce

I have followed all the steps correctly in setting up Genmon. I have checked every config file I can and changed the Serial Port to /dev/ttyAMA0. It still seems that Genmon is attempting to use Serial0.

### Expected Result

Settings to change and start attempting to use /dev/ttyAMA0 to communicate with the generator.

### Actual Result

myserial.log

2025-04-03 13:42:41,444 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 13:42:41,445 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 13:42:41,445 : Error on serial port init!
2025-04-03 14:26:23,640 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 14:26:23,640 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 14:26:23,640 : Error on serial port init!
2025-04-03 14:33:02,977 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 14:33:02,977 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 14:33:02,977 : Error on serial port init!
2025-04-03 14:55:59,401 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 14:55:59,401 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 14:55:59,401 : Error on serial port init!
2025-04-03 14:59:39,681 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 14:59:39,682 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 14:59:39,682 : Error on serial port init!
2025-04-03 15:03:26,758 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 15:03:26,759 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 15:03:26,759 : Error on serial port init!
2025-04-03 15:04:08,102 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 15:04:08,103 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 15:04:08,103 : Error on serial port init!
2025-04-03 15:06:43,752 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 15:06:43,752 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 15:06:43,752 : Error on serial port init!
2025-04-03 15:10:21,840 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-03 15:10:21,840 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-03 15:10:21,840 : Error on serial port init!
2025-04-04 07:31:51,757 : Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
2025-04-04 07:31:51,758 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126
2025-04-04 07:31:51,758 : Error on serial port init!

### Version Info

Genmon Version: 1.19.05

### Logs

[myserial.log](https://github.com/user-attachments/files/19685415/myserial.log)

jgyates, 2025-04-10:
Did you change the file /etc/genmon/genmon.conf? This is the file that genmon uses. Changed the file in your home folder does nothing and is only a copy of the orginal unaltered conf file.

jgyates, 2025-04-10:
Can you provide more info on your installation? Are you using a raspberry pi? What OS? We would need to drill down to determine what is different about your installation as this functionality has been working for some time and the code related to this has not changed in a while. 

I validated and I am able to change the serial port from /dev/serial0 to /dev/ttyAMA0 on my pi

Does the web interface load? What settings are working? How are you validating they are working? Do you see any errors in /var/log/genmon.log?

ConnorMX25, 2025-04-10:
I am using a Raspberry Pi 4 Model B with Raspberry Pi OS (x64)

The Web Interface loads just fine and the name of the site setting is working correctly. The weather settings are working correctly. I haven't really changed any other settings as I wanted to establish communication with the generator first and make sure that my cable was working properly. I am using this converter https://www.amazon.com/HiLetgo-MAX3232-Converter-Module-Serial/dp/B00LPK0Z9A?pd_rd_w=GVmuZ&content-id=amzn1.sym.cd152278-debd-42b9-91b9-6f271389fda7&pf_rd_p=cd152278-debd-42b9-91b9-6f271389fda7&pf_rd_r=5VPQYSQ2DD9GFQVKAHH5&pd_rd_wg=qP1YP&pd_rd_r=fbe6e7b1-cf5f-4454-8201-5378a5898ec1&pd_rd_i=B00LPK0Z9A

I am new to Raspberry Pi so is it possible to just associate port ttyAMA0 with serial0 to allow for the software to run? This Pi will soley be used for monitoring the generator and never have any other software installed on it.

jgyates, 2025-04-10:
The log you send shows errors about not finding /etc/serial0 from April 4th. Are new log entries being added every time you start genmon and your system time is incorrect, or are these old entries from April 4th in your myserial.log file?

If you web interface is up, my guess is that genmon is picking up /dev/ttyAMA0 from your /etc/genmon/genmon.conf file with no issues. If you are successfule then there would be no new entries in the /var/log/myserial.log file.

You can look at the monitor page on the web interface under "Communication Status" and validate that you serial port is working by looking at the communication stats packet count numbers (M for Master or Raspberry Pi, S for Slave or Controller)

ConnorMX25, 2025-04-10:
When I test using the serial test it passes. Could the generator controller be the issue?

[image]

jgyates, 2025-04-10:
No.

the file /var/log/myserial.log has time stamps for each line in the log:

2025-04-04 07:31:51,758 : Error in init: Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0' : myserial.py:126

The 2025-04-04 in the line indicates this an error from 6 days ago. I suggest you restart genmon and see if you see lines in the file with the date of today. 

The log files (like /var/log/myserial.log) do not get deleted with genmon restarts so you are seeing errors from days ago, not today. It appears you genmon is working unless you can provide other data to indicate that is is not.

If your outbound email is setup and working you can submit your log files to me on the About page if you still think you have a serial communications issue.

ConnorMX25, 2025-04-10:
It seems that the 4th is the last day it logged anything. I restarted Genmon and then checked the logs again by downloading them via the web interface and also going into the /var/log/ folder and checking and they are not receiving any logs from today. The web interface is up and I do believe that the software is running I have lines that have been logged for today in my genmon.log file but nothing in the myserial.log.

jgyates, 2025-04-10:
Ok. Sounds like everything is working. I am going to close this thread but feel free to post any updates here.

closed 2025-04-10 by jgyates