jgyates/genmon #1278: WiFi not present when using USB WiFi

skipfire, 2025-04-20:
### Steps to Reproduce

Connect a USB WiFi to the Pi
Configure WiFi connection for the USB dongle
Remove WiFi configuration for the onboard module
Reboot just to be sure

### Expected Result

WiFi strength would show

### Actual Result

WiFi strength doesn't show, it is enabled in advanced settings.

I've been trying to figure out why it doesn't work without success. When booting with a usb wifi the dongle is wlan0 and the onboard is wlan1. The command to find the adapter seems to work (as long as Ethernet isn't also connected)
`ip link | grep BROADCAST | grep -v NO-CARRIER | grep -m 1 LOWER_UP  | awk -F'[:. ]' '{print $3}'`
returned `wlan0` as I would expect.
If you wanted to have that return an adapter even with Ethernet another `grep wlan` I think would solve that.

### Version Info

V1.19.05

### Logs

Submitted, hopefully received.

jgyates, 2025-04-20:
Hi @skipfire 

The line "ip link | grep BROADCAST | grep -v NO-CARRIER | grep -m 1 LOWER_UP  | awk -F'[:. ]' '{print $3}'" will return the first adapter that is "connected" and configured via IP regardless if it is Ethernet or wlan. If I changted it to return a wlan adpater even if Ethernet was configured then no Ethernet adpater would show up.

Genmon assumes one network adapter is active to show the active network connection. if the first network adapter stars with "wl" then it assumes it is wi-fi based. If you have two active it will only show the first one.

 If you have an Ethernet cable connected the wifi will not show up. Let me know if this is not what you are seeting with the USB adapter.

skipfire, 2025-04-20:
Ok, it looks like it's just an issue with the gauge, I found the data on the Monitor page. I do have CPU & WiFi turned on in advanced settings and CPU shows up.  I just rebooted my 3B+ with the USB wifi and no ethernet to ensure nothing was cached from boot, and it doesn't show the gauge but does have the details on the Monitor page.

jgyates, 2025-04-20:
what is the output of this command:

    iw wlan0 link

jgyates, 2025-04-20:
also show the output of this command:

   cat  /proc/net/wireless

skipfire, 2025-04-20:
I threw in an iwconfig wlan0 for reference once I saw that iw link said not connected which is not correct.
```
genmonpi@genmon:~ $ iwconfig wlan0
wlan0     IEEE 802.11AC  ESSID:"psmewap01"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.18 GHz  Access Point: 76:FE:CE:83:FF:EE
          Bit Rate:434 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=100/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

genmonpi@genmon:~ $ iw wlan0 link
Not connected.
genmonpi@genmon:~ $ cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000  100.  100.    0.       0      0      0      0      0        0
genmonpi@genmon:~ $
```

skipfire, 2025-04-20:
And with that info, ChatGPT says that iw relies upon nl80211 while a bunch of Linux drivers, including rtl8812au don't have it fully implemented but older tools like iwconfig use ioctl interfaces instead. So I guess it's a driver issue.

jgyates, 2025-04-20:
I checked in a small update that may help with this. Let me know how your testing goes.

jgyates, 2025-04-20:
This issue, I think, is that the Monitor page get the signal strength by doing this:

     iw wlan0 link

If that fails then it parses /proc/net/wireless.

The gauge only tried "iw wlan0 link" then stopped. I updated it to make the gauge parse /proc/net/wireless as a fallback

you are right that the driver controls the format of this output to a some degree. This is the reason that genmon uses 2 methods.

skipfire, 2025-04-20:
The gauge does show, but pinned at -100dBm.  It looks like `/proc/net/wireless` reports 2 different things depending on what's reporting, probably just logging whatever the driver puts out.  The onboard seems to give dBm, while the USB gives strength.
```
genmonpi@genmon:~ $ cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000  100.   71.    0.       0      0      0      0      0        0
 wlan1: 0000   66.  -44.  -256        0      0      0      0      0        0
```

jgyates, 2025-04-20:
OK. Good to know the limits. Maybe just disable the gauge for USB one. I am guessing it will vary depending on the underlying driver.

skipfire, 2025-04-20:
> OK. Good to know the limits. Maybe just disable the gauge for USB one. I am guessing it will vary depending on the underlying driver.

and that would just take undoing that last change.

closed 2025-04-20 by skipfire

jgyates, 2025-04-21:
I am tempted to leave the update in as this would allow it to work on some of the other drivers that did report dBM. You can still disable the gauge on the advanced settings page. The last update makes the logic on getting the signal strength on the Monitor page stats the same as the gauge.

skipfire, 2025-04-21:
Would an alternate gauge based on percentage signal strength be problematic? If it's positive it uses a percent gauge and negative uses the dBm gauge? Maybe 100-80 green, 60 to 80 yellow, <60 red?  Just spitballing.

jgyates, 2025-04-22:
Thanks, I will look into this

skipfire, 2025-05-02:
Awesome, I'll give it a shot this weekend!

jgyates, 2025-05-05:
oops, try it now. I just checked in an update that should fix it.

skipfire, 2025-05-05:
That is fantastic, thanks!