jgyates/genmon #1279: Fuel type identified incorrectly

lmackley, 2025-04-28:
### Steps to Reproduce

Install genmon.
View Maintenance tab.

### Expected Result

Fuel Type : Natural Gas

### Actual Result

Fuel Type : Propane

### Version Info

V1.19.05

### Logs

Controller is: Evolution, Air Cooled
Fuel is: Natural Gas
Checked setting in controller and it is identified as NG.
This used to work but I had to rebuild my Raspberry Pi from scratch and now it identifies incorrectly.

jgyates, 2025-04-28:
Evolution 1 or 2?

I would need you registers to attempt to replicate this. If your outbound email is setup and working you ca submit you logs on the About page.

jgyates, 2025-04-28:
I refreshed my memory by looking at the code. The Evo Air cooled models do not report the fuel type via modbus. The liquid cooled models do report this via modbus. To let genmon know the fuel type you must set it on the Settings page. Genmon uses this for fuel consumption calculations,

closed 2025-04-28 by jgyates

lmackley, 2025-04-28:
I must have changed the setting before and forgot that I did.
Thanks so much.