jgyates/genmon #1281: Unable to clear "Inspect Battery" alarm from GenMon maintenance screen

mshepard00, 2025-04-29:
### Steps to Reproduce

Generac 22kW air cooled with Evolution 2 controller, GenMon v1.19.05
While onsite and installing GenMon controller 2 weeks ago I saw that battery maintenance interval was approaching and did required battery  maintenance, made entry in Service Journal.  When Date for battery maintenance rolled around Service Log shows "Inspect Battery" and GenMon left side menu shows red instead of green.  Evolution controller display shows "Ready to Run/Inspect Battery" and external monitor light is green.  Generator can be started remotely from Maintenance page but "Reset Alarm" function will not clear the status message "System in Alarm : Check Battery" or turn left menu green.  I assume if I try to reset the alarm on the controller it will succeed, but I am leaving it as is for now for troubleshooting.
I am typically 500 miles away from site and would be confident in clearing this routine maintenance reminder remotely as the periodic maintenance was performed shortly before the timeout.
I am unsure if I am missing a step or if this is an actual issue.

### Expected Result

Clear "Check Battery" alarm, turn left menu highlight from red to green

### Actual Result

Does not clear "Check Battery" alarm and turn left menu highlight from red to green

### Version Info

1.19.05

### Logs

Logs and registers submitted from "About" page

[genmon_logs.tar.gz](https://github.com/user-attachments/files/19964842/genmon_logs.tar.gz)

mshepard00, 2025-04-30:
Thank for your quick response.

I was able to execute the resetmaintalarm command but unfortunately it did not change the state of the red menu or the system alarm state.

login as: genmonpi
genmonpi@192.168.10.20's password:
Linux genmon 6.6.74+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.74-1+rpt1 (2025-01-27) armv                               7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Apr 13 17:56:25 2025 from 192.168.10.1467
genmonpi@genmon:~ $ cd genmon
genmonpi@genmon:~/genmon $ python3 ./ClientInterface.py
CRITICAL: System in alarm! : System in Alarm, Off - Ready
>generator: setremote=resetmaintalarm
Remote command sent successfully
>exit
genmonpi@genmon:~/genmon $ python3 ./ClientInterface.py
CRITICAL: System in alarm! : System in Alarm, Off - Ready
>exit
genmonpi@genmon:~/genmon $

jgyates, 2025-04-30:
ok. Thanks for testing this. It looks like this command is not supported on the Evolution air cooled. The only way to reset this is from the controller.

One more question, what is your controller firmware version? You can see this on the Maintenance page.

jgyates, 2025-04-30:
FYI, this is where it was tested on an evolution 1.0: https://github.com/jgyates/genmon/issues/946

closed 2025-04-30 by jgyates

mshepard00, 2025-04-30:
From Maintenance screen:

 Controller Detected : Evolution 2.0, Air Cooled
 Hardware Version : V1.00
  Firmware Version : V1.18

Performed local reset at generator.  Controller prompted if maintenance was performed

genmonpi@genmon:~/genmon $ python3 ./ClientInterface.py
CRITICAL: System in alarm! : System in Alarm, Off - Ready
>exit
genmonpi@genmon:~/genmon $ #Local reset OFF-ENTER-"Maint Performed?"-ENTER-AUTO
genmonpi@genmon:~/genmon $ python3 ./ClientInterface.py
OK : Auto, Off - Ready
>exit
genmonpi@genmon:~/genmon $