jgyates/genmon #1284: tankutility is not updating

capindc2, 2025-05-05:
If I long in to tankutility.com the redirect is app.generacfuel.com  last Last Reading Date Apr 24, 2025 3:56:31 PM.  If I log in to app.mobilelinkgen.com Last Reading Date: 05/04/2025 10:02 PM EDT 

Battery is reporting Good, If I restart I get green lights no red lights

Any help

jgyates, 2025-05-05:
Hi @capindc2 

Unfortunately this is out of the control of the genmon add on. All genmon does is read the data from the cloud. Everything else is controlled by the tanktutility firmware and cloud.

I can add some data to the mix. I have a tankutility device that I use on a propane tank that is not used by my generator. My device was working fine for a few years but I noticed that it gradually started have a delay between the reading time and when the reading was available to read over the API. I also noticed some differences between the two sites. My guess is that one of the sites is updates from the deice and the other is copied for legacy software support. I am guessing the copy is delayed. 

Then I noticed the actual readings were being skipped (they used to be every day). It eventually stoped doing daily reads and went to about once a week. Now my device has not updated since 4/8/2025. I tried new batteries to no avail. I have not tried to look into this in-depth, but if it watch the lights the lights indicate that the reading is never uploaded to the cloud. My device is at the far end of an acceptable wifi signal so a guess is that my wifi signal is degraded due to corrosion on the internal tankutiliy antenna. I noticed here was some slight corrosion on the internal parts when I changed the battery.

capindc2, 2025-05-05:
Thanks for the insight this is a newish cell service device like. I think there is something going on with the Tankutility service sent email it bounced. call number left VM  I will keep you posted. 

Does it still look at https://app.tankutility.com for it log in

capindc2, 2025-05-09:
More information I have the Generac 4G LTE tank reader, I am part of a Home Assistant group there is an Integration for Generac genset so you can add it to HA it looks like Generac MobileLink got and an upgrade V3.8 and it's possible that broke Tankutility because it pull data from the same API. It broke the HA Integration service. I have reach out to Tank Utility tell no reply.

bgiuriceo, 2025-05-09:
Same here. I called on wednesday and they had no idea. Also they have changed monitoring plans that could not be purchased. I opened a ticket and am awaiting resolution. (Ecobee, Generac and Tank Utility).

capindc2, 2025-05-09:
> Same here. I called on wednesday and they had no idea. Also they have changed monitoring plans that could not be purchased. I opened a ticket and am awaiting resolution. (Ecobee, Generac and Tank Utility).
> […](#)

How did you open a ticket the email bounced and the phone message said please txt this number for faster reply.

bgiuriceo, 2025-05-09:
> … <x-msg://5/#>

I called the support # on the mobile link webpage. I spoke to Mike and he relayed it to IT. Right now it seems that billing has been worked out. If you don’t pay your tank monitor will show no signal. You have to pay going forward.

capindc2, 2025-05-09:
Oh app.mobilelinkgen.com (generac) is not the same as Tank utility app.generacfuel.com log in to generacfuel.com (tankutility)  see when the last time your tank was updated.

bgiuriceo, 2025-05-09:
The ‘old’ thank utility no longer works I access everything through the Generac Web app. Everything seems to be working now. It also appears that tank readings are made and presented to my ecobee, but not presented in the web app. It shows the tank having no RF reception. I hope this helps but everything is right once you pay the ransom. There was also a problem with posting to a credit card and getting a prorated payment that all seems to have been worked out now also, I hope I'm answering your question properly.

capindc2, 2025-05-09:
@jgyates if you want to close this I understand, I also understand more of what is going on a number of years ago Tank utility was bought by Generac and is slowly dying or being phased out I talked Matt in the Tank Utility division of Generac  he said he can look in to having my device ID moved back to the old API as long as I am ok with lower security. There is a credit card on file. but no Genset access only Tank reading. I said I was fine with that, if this does work I will ask to share he email if anyone else wants to have there device moved to the old api.

closed 2025-05-09 by jgyates

capindc2, 2025-05-11:
@jgyates Right now the guys on the home assistant add-on are working on fixing the login problem give the right info could you build an add-on that would login  right in to Generac mobile link website to pull the data for the tank from there. If you need login info I can send you mine.

I guess there is 2 guys working on the same project here are the 2 link I hope this help.

binarydev
https://github.com/binarydev/ha-generac/issues/183

bentekkie
https://github.com/bentekkie/ha-generac/issues/139#issuecomment-2869294056

jgyates, 2025-05-11:
yes, I should be able to modify the existing add on to have an option to work the new way.

binarydev, 2025-05-11:
Hey all happy to collaborate if there's any need or interest. I'm away on vacation, but will be getting a fix out in my linked repo (binarydev/ha-generac) above when I return this coming weekend (unless someone is able to test the proposed fix before then for me to accept). My repo above also supports Generac propane tank monitoring, which the second repo does not (my repo is a fork of the less maintained repo from bentekkie).

binarydev, 2025-05-12:
Thanks to some wonderful community members, a fix is now available in my repo and in the HACS index via the "generac" integration v0.3.6, as noted in https://github.com/binarydev/ha-generac/issues/183#issuecomment-2870457085

jgyates, 2025-05-12:
The genmon add on uses the older https://app.tankutility.com/  URL to log in and retrieve the data. I brought my unit inside as I think my problem is due to decreased wifi performance. I forced a reading on the device and in a short period of time the reading is showing up on both https://app.generacfuel.com and https://app.mobilelinkgen.com/

My unit is a wifi based unit that was purchased prior to the generac buyout. Is this issue that 4G based devices only work with the new login? Just tring to understand what is and is not working before I make any changes.

jgyates, 2025-05-12:
and to clarify, the HA plug in uses the newer generac URL for logging in (https://app.mobilelinkgen.com/api), genmon uses  https://app.tankutility.com/, which appears to still be working.

capindc2, 2025-05-15:
@jgyates  Thank you
 I still tried to make this work for the life of me I can not get Tankutility  (https://app.tankutility.com/) to update. I have tried  everything from deleting the devices on both Tankutility and Mobile link. The only thing I can think of is I have the Generac LTE Kit Tank Monitor and Tankutility does not have access to it. So for 50 bucks I bought a Wifi ver. to see if that work.

jgyates, 2025-06-12:
Sure, I can add this to the list of things to do. I will not be able to dedicate any significant time to it for about 3 week.

capindc2, 2025-06-13:
Thank you Sir
no rush

AGTrifecta, 2025-06-14:
You can still log into the Tank Utility website/app, but my sensor has not updated it for several months. A call to Generac found that they've moved things to app.mobilelinkgen.com. I had to create a new account, reset the sensor, and pay a subscription for access. As noted above, the add-on probably needs to start talking to mobilelinkgen.com.

capindc2, 2025-06-26:
> You can still log into the Tank Utility website/app, but my sensor has not updated it for several months. A call to Generac found that they've moved things to app.mobilelinkgen.com. I had to create [...]

If you are using the LTE Ver. Tank Utility is no longer supporting the model @jgyates  is going to add an Add-on to use the API from app.mobilelinkgen.com