jgyates/genmon #1286: Did an upgrade to the latest version,  Now failing to start -"Enable to find config file /home/genmonpi/genmon/conf/gencentriconnect.conf"

markboyd-info, 2025-05-13:
### Steps to Reproduce

I clicked the button in the Genmon application to upgrade to the latest version.  Now Genmon fails to start with the following:
./startgenmon.sh start
using binary: /home/genmonpi/genmon/genenv/bin/python
Starting genmon python scripts
Enable to find config file /home/genmonpi/genmon/conf/gencentriconnect.conf
Error validating config. Retrying..
Running one time maintenance check
Enable to find config file /home/genmonpi/genmon/conf/gencentriconnect.conf
Error validating config file, Exiting 

The /home/genmonpi/genmon/conf/gencentriconnect.conf file does not exist.   

### Expected Result

When I click the button to upgrade, I expected it to handle the upgrade and restart Genmon with the latest version.

### Actual Result

Genmon fails to start saying its missing the  /home/genmonpi/genmon/conf/gencentriconnect.conf
which indeed does not exist. 

### Version Info

Upgraded to the latest version.
Running:
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm

jgyates, 2025-05-13:
Sorry about that. About an hour ago I checked in what I though was a harmless update that goes along with some upcoming changes. Turns out it was not harmless. I just checked in an update that should fix this. You can update without genmon loaded opening a terminal window (via ssh or locally) and changing directories into the genmon folder:

     cd genmon

Then type this command:

     ./genmonmaint.sh -u

Then press 'y' to update the software. Then you can either reboot the pi or type this to resstart genmon

     ./startgenmon.sh restart

Let me know if you have any issues.

jgyates, 2025-05-13:
I checked the latest update and it is working as expected now.

closed 2025-05-13 by jgyates