jgyates/genmon #1297: Power output value incorrect

griffths, 2025-07-02:
### Steps to Reproduce

[image]

### Expected Result

correct output values

### Actual Result

My installation of Genmon had been running without issue for quite a while and I happened to hear the generator startup for its weekly exercise so I loaded Genmon to see what it was doing and was surprised to see the power output values off the charts.  The genset was not supplying any power to the house at the time of the screenshot.

I have been keeping up with the releases and seem to recall something was done recently to correct the output graphs for certain gensets and am wondering if that has affected my installation of Genmon

### Version Info

I am running V1.19.06

### Logs

Logs and registers submitted via about page.

jgyates, 2025-07-02:
Hi @griffths 

Based on your logs, you are using a Nexus liquid cooled unit, which does not current readings butilt into the controller. It also appears you are using external current transformers (CTs) with the external CT Hat add on. 

If your readings for Current leg 1 and 2 are out of range, then this would cause the power display to be invalid. 

I good first step would be to validate the CTs are outputing valid data because they are showing more current that your generator supports in the web interface. To do this:

1) edit the file /etc/genmon/gencthat.conf an add the line to the bottom of the file if it does not already exist:

debug=True

save the file an restart genmon. Then start the generator and let it run until you get readings on the web interface for current leg 1 and 2. Then stop the generator an resubmit your log files. You can view the additional data collected in the file /var/log/gencthat.log

If the current reading form the add on log file are out of range then this the source of your problem. 

Let me know how it goes.

griffths, 2025-07-03:
@jgyates ,

Thank you for your reply with troubleshooting steps.

I logged into Genmon and the gauges are now reading zero.  The genset has not run since my last email and I have not taken any of the steps in your detailed response.

I just manually started the genset without a transfer and the power output is staying at zero.

We did have a very short power blip so maybe that glitched the software or Pi which eventually corrected itself.

Out of curiosity, I will perform your steps but I suspect this issue was a red herring.

Regards,

Jay

jgyates, 2025-07-03:
OK. I will close this issue but feel free to post any follow up to this thread.

closed 2025-07-03 by jgyates

griffths, 2025-07-04:
Enabled debug and rebooted the Pi but now the power graph and gauges are missing.

I tried a power cycle and same result.

At the moment, I am going backwards.....

[image]

jgyates, 2025-07-04:
The logs indicate that it is only reading zeros from the CTs. Channel 0 is the 1st CT, Channel 1 is the second. num samples - the number of reads, max is the max value read, min is the minumum value read. ms elapsed is the milliseconds that elapse during the number of attempted reads.

Of course these numbers could be zero depending on where you have attached the CTs and if the generator is running. 

How are the CTs connected?